Spritetm/hadbadge2019_fpgasoc #6: Cannot compile soc

szczys, 2019-10-04:
Why typing `make` in the soc folder I'm first hit with an error about unconstrained pins.
```
Using template $paramod\$lut\WIDTH=4\LUT=16'0111000100000000 for cells of type $lut.
Using template $paramod\$lut\WIDTH=4\LUT=16'0000000011010111 for cells of type $lut.
No more expansions possible.
<suppressed ~20545 debug messages>
Removed 0 unused cells and 21554 unused wires.

45.31. Executing HIERARCHY pass (managing design hierarchy).

45.31.1. Analyzing design hierarchy..
Top module:  \top_fpga

45.31.2. Analyzing design hierarchy..
Top module:  \top_fpga
Removed 0 unused modules.

45.32. Printing statistics.

=== top_fpga ===

   Number of wires:              26757
   Number of wire bits:          45012
   Number of public wires:        1575
   Number of public wire bits:   14029
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              28388
     CCU2C                        1128
     DP16KD                         32
     EHXPLLL                         2
     JTAGG                           1
     L6MUX21                      1630
     LUT4                        16441
     MULT18X18D                      8
     ODDRX1F                         8
     OSCG                            1
     PFUMX                        4034
     TRELLIS_DPR16X4               555
     TRELLIS_FF                   4485
     TRELLIS_IO                     62
     USRMCLK                         1

45.33. Executing CHECK pass (checking for obvious problems).
checking module top_fpga..
found and reported 0 problems.

45.34. Executing BLIF backend.

45.35. Executing JSON backend.

Warnings: 23 unique messages, 23 total
End of script. Logfile hash: 63e3f25c68
CPU: user 88.84s system 0.61s, MEM: 1302.07 MB total, 1272.42 MB resident
Yosys 0.9+231 (git sha1 fda94311, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)
Time spent: 25% 7x techmap (23 sec), 18% 22x opt_clean (16 sec), ...
nextpnr-ecp5 --json soc.json --lpf had19_proto2.lpf --textcfg soc_out_synth.config --45k --package CABGA381 --speed 8  --pre-pack clock-constrainsts.py --seed 2
Info: Importing module top_fpga
Info: Rule checker, verifying imported design
Info: Checksum: 0xd290047b

ERROR: IO 'sao1[0]' is unconstrained in LPF (override this error with --lpf-allow-unconstrained)
0 warnings, 1 error
Makefile:45: recipe for target 'soc_out_synth.config' failed
make: *** [soc_out_synth.config] Error 255
```

If I add the suggested --lpf-allow-unconstrained flag to the makefile:
```
--- a/soc/Makefile
+++ b/soc/Makefile
@@ -42,7 +42,7 @@ $(PROJ).json $(PROJ).blif: $(SRC) $(SRC_SYNTH) $(EXTRA_DEPEND)
                          synth_ecp5 -top top_fpga -json $(PROJ).json -blif $(PROJ).blif"
 
 %_out_synth.config: %.json clock-constrainsts.py
-       nextpnr-ecp5 --json $< --lpf $(CONSTR) --textcfg $@ --45k --package CABGA381 --speed 8  --pre-pack clock-constrainsts.py --seed 2
+       nextpnr-ecp5 --lpf-allow-unconstrained --json $< --lpf $(CONSTR) --textcfg $@ --45k --package CABGA381 --speed 8  --pre-pack clock-constrainsts.py --seed 2
 
 %_out.config: %_out_synth.config bram_replace/bram_replace$(EXE) rom.hex
        bram_replace/bram_replace$(EXE) -l 8192 -r $< $@ rom.hex
```
I still cannot build the soc:
```
ERROR: Max frequency for clock                  '$glbnet$clk48m': 49.34 MHz (FAIL at 50.00 MHz)
Info: Max frequency for clock            '$glbnet$vid_pixelclk': 68.47 MHz (PASS at 30.00 MHz)
Info: Max frequency for clock                  '$glbnet$clkint': 279.25 MHz (PASS at 25.00 MHz)
Info: Max frequency for clock '$glbnet$hdmi_encoder.clk_125MHz': 343.52 MHz (PASS at 140.02 MHz)
```

Spritetm, 2019-10-04:
The 'unconstrained soc' bit is weird, I'll look into that. The timing fail is known; it is kind-of random (as the placer uses a random-ish scheme to place the components) and sometimes can be worked around by hacking the Makefile to use a different seed, but in the current master it is somewhat prevalent. I've optimized the cache a bit in the gfx_hacky_hacky branch; I'll merge that as soon as I have the issues there (mostly cache flush not actually flushing) ironed out.

Spritetm, 2019-10-04:
Should be working now: I merged the gfx_hacky_hacky branch that has better timing and fixed the proto2 lpfs so they should work with the current verilog. Please pull and retry if you want.

commit 84826fa9 (2019-10-04): Sync proto2 and proto3 lpfs. Fixes #6
closed 2019-10-04 by Spritetm