Spritetm/hadbadge2019_fpgasoc #9: Unable to flash helloworld app

szczys, 2019-10-04:
With the closing of #7 I can successfully compile the helloworld app into an .elf file. With the most up-to-date SoC and IPL the badge comes to life and it mounts as mass storage. I have copied the elf file over to it, but I'm not sure how to then flash (run?) the app. The file is persistent between power cycles, but it doesn't seem to actually run on the badge.

szczys, 2019-10-04:
Further investigation into the IPL code has me realizing that the far right button can load the app, but it's looking for a static filename of `autoexec.elf`:
```
Loading app autoexec.elfautoexec.elf: No such file or directory
Loading app autoexec.elfautoexec.elf: No such file or directory
```
If I rename helloworld.elf accordingly the app runs successfully, although all it does it printf to the serial terminal:
```
Go!
Hello from app!
opening
hello.txt: Success
autoexec.elf
End.Go!
Hello from app!
opening
hello.txt: Success
autoexec.elf
End.
```

App running workflow:
------------------------------

1. Compile app to elf using `make`
2. Rename app to `autoexec.elf`
3. With badge turned on, connect it to computer via USB
4. Copy autoexec.elf to badge mass storage folder
5. Eject mass storage from your system
6. App can be run with right button without need to powercycle

davidthings, 2019-10-13:
Is this possibly windows (or something else) dropping an autoexec.something on the drive?  Then the badge directory scanning code finds it?  Under Linux, building app-helloworld results in helloworld.elf.  Dragging that file to the board's drive makes it pop up on the menu.  It can be run from there.

Spritetm, 2019-10-14:
Yeah, closing this comment as it was on a version before the menu allowing you to select an app.

closed 2019-10-14 by Spritetm