Spritetm/hadbadge2019_fpgasoc #11: Need toolchain and compilation instructions

szczys, 2019-10-11:
I've put together the steps I've used sucessfully here:
https://gist.github.com/szczys/f4ec190bfe5ceaf851ca03e4f9090393

Not working on one of my machines is the tinyprog. It seem the pip3 repo install method should work but so far I get a device or resource busy on ttyACM0 that I haven't been able to work out yet. 

I'll keep working on it but wanted to open this issue in case anyone else was also snooping around on this topic.

smunaut, 2019-10-11:
I would actually recommend switching over to the https://gnu-mcu-eclipse.github.io/blog/2019/05/21/riscv-none-gcc-v8-2-0-2-2-20190521-released/ toolchain.

The one from sifive is mostly built for large riscv with linux support and such. This one is targeted at much smaller targets and include specs file to reflect that. (nano.specs for instance).

davidthings, 2019-10-13:
The riscv32-unknown-elf- toolchains are deprecated.  New ones are here:

https://xpack.github.io/riscv-none-embed-gcc/install/#manual-install

The riscv-none-embed- tools work fine once the Makefiles (in apps-sdk (sdk.mk), boot, & ipl) reflect the new preffix:

eg

```
#PREFIX := $(RISCV_TOOLCHAIN_PATH)/riscv32-unknown-elf-
PREFIX := $(RISCV_TOOLCHAIN_PATH)/riscv-none-embed-
```

szczys, 2019-10-14:
@xobs do you have any thoughts on this?

smunaut, 2019-10-14:
@davidthings Oh yeah, I hadn't noticed that the gnu eclipse ones had been moved over to the xpack project.

xobs, 2019-10-14:
@szczys I was planning on moving to this toolchain once the gdb bug I reported was fixed. That issue was closed yesterday, so I'll take another look.

I was also investigating the possibility of switching packaging methods to allow me to include tinyprog, which should solve your other issue.

Spritetm, 2019-11-11:
I'm gonna close this as the docs have a nice place for toolchain instructions now.

closed 2019-11-11 by Spritetm