Spritetm/hadbadge2019_fpgasoc #13: Workarounds required for tinyprog on Ubuntu 18.04

szczys, 2019-10-14:
I'm running Mint 19 which is effectively Ubuntu 18.04. I have tinyprog working fine on this system on my laptop, but my desktop doesn't want to play nicely with tinyprog. I get the following error despite being in the dialout group and having permissions:
```[Errno 16] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'```

After much mucking about I believe I've found a workaround. Here's the workflow I followed:

## Install tinyprog

1. Clone the badge repository: `git clone https://github.com/Spritetm/hadbadge2019_fpgasoc.git --recursive`
2. `cd hadbadge2019_fpgasoc/TinyFPGA-Bootloader/programmer/`
3. `sudo python3 setup.py install`

## Use workaround if necessary

If you get the Device or resource busy error when doing a `make flash` try this and then run the flash command again:
` sudo systemctl stop ModemManager.service`

Not sure why this fixes it , I found it here: https://discourse.tinyfpga.com/t/new-tinyprog-problems-under-ubuntu/1148

davidthings, 2019-10-14:
Mike this is a well known problem under Mac and Linux.  TinyProg Bootloader (and many other USB implementations) advertise that they're Modems, so the operating system obliges and fires up the Modem Manager.  Of course, a modern person doesn't have much to do with POTs Modems, so this function can be safely disabled.

smunaut, 2019-10-15:
One more reason to switch to DFU :D

smunaut, 2019-11-12:
I think we can close this ?

szczys, 2019-11-12:
Yes, I think this is a non-issue at this point.

closed 2019-11-12 by szczys