Spritetm/hadbadge2019_fpgasoc #18: Pmod Connector Labels Reversed

davidthings, 2019-10-16:
Small issue... The schematic has the pinout for the Pmod wrong.  A surprising claim for the sloppiest [standard](http://digilentinc.com/Pmods/Digilent-Pmod_%20Interface_Specification.pdf) I've ever seen.  

Slop not withstanding, everything's very consistent on one point: Pin 1 is on the *opposite* end from the supply lines, and the numbering is as if two 6-pin Pmods are sitting next to each other.

| PMOD | Function |
| -    | -        |
| 1    | Pmod.1    |
| 2    | Pmod.2     |
| 3    | Pmod.3  |
| 4    | Pmod.4  |
| 5    | GND  |
| 6    | VCC  |
| 7    | Pmod.7   |
| 8    | Pmod.8 |
| 9    | Pmod.9   |
| 10   | Pmod.10   |
| 11    | GND  |
| 12  | VCC  |

This excerpt from the LPF file shows the correct way, I think.

``` TCL
LOCATE COMP "pmod[0]" SITE "B13" ;#a1
LOCATE COMP "pmod[1]" SITE "A13" ;#a2
LOCATE COMP "pmod[2]" SITE "C15" ;#a3
LOCATE COMP "pmod[3]" SITE "B15" ;#a4
LOCATE COMP "pmod[4]" SITE "D16" ;#b1
LOCATE COMP "pmod[5]" SITE "A14" ;#b2
LOCATE COMP "pmod[6]" SITE "C16" ;#b3
LOCATE COMP "pmod[7]" SITE "A15" ;#b4
# These Pmod assignments are incorrect
# LOCATE COMP "pmod[0]" SITE "A15" ;#a1
# LOCATE COMP "pmod[1]" SITE "C16" ;#a2
# LOCATE COMP "pmod[2]" SITE "A14" ;#a3
# LOCATE COMP "pmod[3]" SITE "D16" ;#a4
# LOCATE COMP "pmod[4]" SITE "B15" ;#b1
# LOCATE COMP "pmod[5]" SITE "C15" ;#b2
# LOCATE COMP "pmod[6]" SITE "A13" ;#b3
# LOCATE COMP "pmod[7]" SITE "B13" ;#b4
```

Spritetm, 2019-10-21:
Ahrg, jeeze, that means it's wrong on the silkscreen of the final version now, I must've been asleep while adding that connector. Makes me wonder what to do: change the lpf and mismatch the silkscreen or declare the pmod as being 'almost pmod'. Leaving this open because I now want to check the other conns as well/

smunaut, 2019-10-21:
I don't see any silk screen at all ? Am I missing something ?

Spritetm, 2019-10-22:
Proto2 does not have silkscreen for this, proto 3 and the final version have the pinouts of various connectors on the siklscreen, including the (wrong) pmod pinout.

szczys, 2019-11-03:
Here is a picture of the silk on the rear side of badge. We can note this quirk in documentation once it's decided what the best resolution is for the issue. 

[image]