Spritetm/hadbadge2019_fpgasoc #20: make app-helloworld fails with link errors

alanvgreen, 2019-10-22:
Steps:
1. $ cd app-helloworld
2. $ make clean; make

Result: Compile fails with link error
Expected result: successful compilation

There are many errors of this form:
    LD helloworld.elf
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-calloc.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'

It seems that the compiler is producing riscv32 .o files, but the .a file is riscv64.

Something that worked: Recompiling the riscv compiler to get a riscv32 toolchain, and changing the makefiles to use riscv32-unknown-elf-* binaries instead of riscv64-unknown-elf binaries results in a successful build.

I think the problem is that, even though the riscv64-unknown-elf-* toolchain can produce and work with 32 bit binaries, the libraries are all 64 bit. I could not work out how to make the riscv64-unknown-elf toolchain work with 32 bit libraries. I think the right fix is probably just a little documentation.

alanvgreen, 2019-10-22:
Attached is the 
[output of the unsuccessful make](https://github.com/Spritetm/hadbadge2019_fpgasoc/files/3757779/out.txt)

smunaut, 2019-10-22:
Where did you get your toolchain from ?

alanvgreen, 2019-10-22:
I built according to the instructions in the README:

> If you want to compile your own RiscV toolchain, see https://github.com/riscv/riscv-gnu-toolchain . Specifically, you want a multilib Newlib toolchain, something that is not in the list. Just invoke these two commands after having fulfilled all prerequisites and you should be good:
> 
> ./configure --prefix=/opt/riscv
> make

unwiredben, 2019-10-22:
I used the prebuilt tools from https://github.com/xobs/ecp5-toolchain/releases and had no problems. I wonder what's different in how xobs did that build vs. doing it manually.

alanvgreen, 2019-10-22:
Yes, the precompiled toolchain seems to work.

closed 2019-10-22 by alanvgreen