Spritetm/hadbadge2019_fpgasoc #21: Signals on GENIO bus while TinyProg active

davidthings, 2019-10-23:
Starting some interfacing, I was very surprised to see signals on the cartridge IO (eg. 48MHz on genio[0], aka GENIO1)

I thought it must be some exotic ECP5 boot mode, but looking further it looks like debug output for TinyProg!

``` verilog
  assign genio[0] = clk_48mhz;
  assign genio[1] = 0;
  assign genio[2] = usb_tx_en?usb_n_tx:usb_n_rx;
  assign genio[3] = usb_tx_en?usb_p_tx:usb_p_rx;
  assign genio[4] = flash_sck;
  assign genio[5] = flash_cs_i;
  assign genio[6] = flash_mosi;
  assign genio[7] = flash_miso;
  assign genio[27:8] = 'h0;
```

Should they be there?  Someone putting hardware on that port might reasonably expect that their lines don't get busy during Tinyprog...

Maybe OK in most cases, but could cause problems...

Spritetm, 2019-10-25:
Ah, these were indeed very useful for me to connect my LA to, but they certainly are not supposed to be there on the final version. We'll probably change to a DFU-based bootloader anyway, so that should nix that issue. Keeping this open as a reminder.

smunaut, 2019-11-12:
Again, this should be irrelevant now.

Spritetm, 2019-11-15:
Closed because we just flashed ~550 badges with the new DFU bootloader, so this indeed is pretty irrelevant.

closed 2019-11-15 by Spritetm