Spritetm/hadbadge2019_fpgasoc #29: License

alanvgreen, 2019-10-23:
What is the feeling about putting  a GPLv3 license on thadbadge2019_fpgasoc?

My interest in having a license is that my employer, while supportive of open source contributions, has processes that are much simpler for me to follow if the repository has a an OSI approved license. I'm guessing that some supercon attendees will be in a similar situation.

IANAL but, looking through the source I see a wide range of licenses, including GPLv2 and LGPLv3 licensed code, so it looks like GPLv3 would be appropriate.

I'm happy to send a patch adding a license notice.

davidthings, 2019-10-24:
Great point.  But why LPGL or GPL?  Why not put a permissive (Apache or MIT) license on those files that are not already licensed so that all this work can find its way into lots of fun stuff, commercial and otherwise.

alanvgreen, 2019-10-25:
Just that parts already appear to be GPL licensed. By design, GPL makes using a non-GPL license difficult. This makes GPL appear to me to be a straight-forward choice.

Though, as a I wrote before, I am not a lawyer.

davidthings, 2019-10-25:
How funny would it be to see two lawyers speculate about FPGA development. "I'm not an engineer, but...[insert most likely wrong assertion here]"

Spritetm, 2019-10-25:
I think GPL has some issues with hardware, as in gateware != software, so I'd prefer something like BSD there. Will add to my sources. For the open-source stuff, I don't quite care, I'm happy whacking GPL on there. Keeping this open as a ToDo.

alanvgreen, 2019-10-26:
Sounds good. Please let me know if I can help.

Spritetm, 2019-10-26:
I added licenses. As I said, it's mixed-license, with bsd for gateware and gplv3 for the code. Closing this for now, if you need any more stuff for OSI red tape, feel free to comment.

closed 2019-10-26 by Spritetm