Spritetm/hadbadge2019_fpgasoc #43: Need hardware timer for C code

szczys, 2019-11-01:
Apps written in C need access to some reliable hardware timer to implement non-blocking delays.

- Could be a 32-bit register that increments every millisecond
- @Spritetm mentioned it might be possible to add an indication of the 60 Hz vblank signal

smunaut, 2019-11-01:
The riscv architecture includes provision for hardware counters, it's as simple as enabling them in the picorv32 CPU.

Having an interrupt on vblank would probably be useful for games in any case ...

Spritetm, 2019-11-02:
There's a vbl counter now that is pretty useful for e.g. waiting x frames, or waiting until you can write data for the next frame. See ipl main.c for an example.

commit 00be84ed (2019-11-02): Add VBlank counter. Closes #43.
closed 2019-11-02 by Spritetm