Spritetm/hadbadge2019_fpgasoc #48: Add Badgetris! as submodule

szczys, 2019-11-03:
I have a stable version of this ready. Is adding this as a submodule the best way to include it in the project?

```git submodule add https://github.com/szczys/badgetris-2019supercon.git app-badgetris```

## Known issues:

* Background image has some flicker in the lower third of it. I'm unsure if this is my code an issue with the IPL.

Spritetm, 2019-11-03:
Sure! Note that the way submodules work is that they're pinned to a certain commit in your badgetris repo; if you do any upgrades on badgetris, you also need to add a push request to update the module here.

Will add the submodule as soon as I'm done hacking here.

Spritetm, 2019-11-04:
Added.

closed 2019-11-04 by Spritetm