Spritetm/hadbadge2019_fpgasoc #59: Few questions about HDMI and power source

goran-mahovlic, 2019-11-05:
Hi to all, 
does someone has HDMI, power and USB schematics?
It really looks very similar to our board.
And if our board was reference we are really fine with that it is board that is opened from beginning.
Only downside is that if you did use our board as reference you could contact us and send us one sample, we would gladly help in making badge even better. This way we see some possible mistakes that we cannot confirm... 
I think if we want opensource hardware along with toolchain to be even more successful we really need collaborate more!
I did try direct contact before opening issues but still no answer...

Spritetm, 2019-11-05:
I have not seen any direct contact attemps; can you email me at jeroen at spritesmods period com?

goran-mahovlic, 2019-11-05:
sure, sorry I have tried on twitter

goran-mahovlic, 2019-11-05:
You have mail...

goran-mahovlic, 2019-11-05:
We can close that one. Just to note here. Our design is on purpose left open from start. And one strong purpose is education. Feel free to lock and use it as much as you want. But good practice is contact and mention as we try to do every time... Some time when you just peek and do not ask -- purpose remains unknown and we all want to speed up our designs and make them better! More good working open boards -- better for opensource community! https://github.com/emard/ulx3s

closed 2019-11-05 by goran-mahovlic