Spritetm/hadbadge2019_fpgasoc #67: Toolchain doesn't work under Windows CMD

alanvgreen, 2019-11-06:
The toolchain doesn't (yet) work under Windows CMD.  Here are two attempts, one without V=1 and one with:

```
C:\Users\Alan\hadbadge\hadbadge2019_fpgasoc\app-helloworld>make
"    " CC crt0.S
The syntax of the command is incorrect.
make: *** [C:\Users\Alan\hadbadge\hadbadge2019_fpgasoc\apps-sdk/sdk.mk:106: C:/Users/Alan/hadbadge/hadbadge2019_fpgasoc/app-helloworld/build/apps-sdk/gloss/crt0.o] Error 1
```

```
C:\Users\Alan\hadbadge\hadbadge2019_fpgasoc\app-helloworld>make V=1
process_begin: CreateProcess(NULL, true CC crt0.S, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [C:\Users\Alan\hadbadge\hadbadge2019_fpgasoc\apps-sdk/sdk.mk:105: C:/Users/Alan/hadbadge/hadbadge2019_fpgasoc/app-helloworld/build/apps-sdk/gloss/crt0.o] Error 2
```

This is using the latest toolchain from https://github.com/xobs/ecp5-toolchain/releases/tag/v1.6.1.

I have set RISCV_TOOLCHAIN_PATH and PATH to point to the toolchain. I have verified that I can run the tools

I've had success building using Windows Subsystem for Linux (WSL).

alanvgreen, 2019-11-06:
It's bed time and I'll dig into this some more tomorrow (the above seems all related to the tracing code), but wondered if any one else was working on Windows.

smunaut, 2019-11-06:
I checked that it built on both WSL and on a MSYS/Cygwin sheel (since you pretty much need git and git-for-windows comes with msys anyway it didn't really seem like asking much).

alanvgreen, 2019-11-06:
Ah. Makes sense. Yes, Git-for-Windows bash shell Just Works for me too.

 I'll write up a "Quick Start" document.

closed 2019-11-07 by alanvgreen