Spritetm/hadbadge2019_fpgasoc #77: README how-to links 404

gojimmypi, 2019-11-10:
The Linux/Mac/Windows links appears to all be 404 in the [How to use](https://github.com/Spritetm/hadbadge2019_fpgasoc#how-to-use) section of the [README.md](https://github.com/Spritetm/hadbadge2019_fpgasoc/blob/master/README.md)

Spritetm, 2019-11-11:
Fixed in #78 I guess :)

closed 2019-11-11 by Spritetm

gojimmypi, 2019-11-12:
yes! :) I had meant to reference the PR in the issue. Thanks for prompt response. Confirmed all working now, and oooh... cool: A docker image one added. Interesting.

btw - @Spritetm I tried to contact you via web form... not sure if you received it? The folks at Radiona asked that I give you one of the bare ULX3S boards this weekend. Interested? Perhaps send me a gmail? (same userid as my GitHub)