Spritetm/hadbadge2019_fpgasoc #119: Windows RiscV development Example Issue.

FoamyGuy, 2019-11-15:
I am admittedly green with RiscV so perhaps I am overlooking something relatively basic, but I am having trouble trying to follow this basic walk-through in the docs: [https://github.com/Spritetm/hadbadge2019_fpgasoc/blob/master/doc/toolchain-win.md](https://github.com/Spritetm/hadbadge2019_fpgasoc/blob/master/doc/toolchain-win.md)

I have modified my main.c file and it seems to have compiled fine using the make command as far as I can tell. I copied the resulting .elf file to the board and I do see it show up in the main menu list. 

I modified main.c as directed in the doc changing the "Hello World!" string value. 

The issue comes when I run the app on the board. The app does launch and seem to run without crashing, but on the screen I am still seeing "Hello World!" as the text on the screen. As far as I can tell my edits to the main.c file are having no effect. The app continues to look exactly like it does when I run the un-modified hello world app. 

Are there any steps required that have been left out of the doc page linked above that I may be missing which could cause my app to essentially ignore my changes to the "Hello World!" String

Spritetm, 2019-11-15:
Yeah, 'Hello World' is a bit misguided... it has some tile loading code as well, but the tiles overwrite anything that is printed on that layer beforehand. Lemme make it a bit more sane real quick.

Spritetm, 2019-11-15:
OK, should be fixed now.

closed 2019-11-15 by Spritetm

FoamyGuy, 2019-11-15:
I see. I was just starting to figuring out that I need to be editing the tilemap file rather than the print commands. Thank you for the reply and updating the sample app to be clearer.