Spritetm/hadbadge2019_fpgasoc #122: PCM click fix causes violation of 48 MHz constraint

ianjfrosst, 2019-11-15:
With the changes in #118, making the `soc/` errors with:
    ERROR: Max frequency for clock                  '$glbnet$clk48m': 46.22 MHz (FAIL at 48.00 MHz)

Spritetm, 2019-11-16:
Could depend on your see, can you change soc/Makefile to try a different one?

ianjfrosst, 2019-11-16:
Tried and found a value that works. Should I make a PR for it?

Spritetm, 2019-11-16:
You can if you want to, but there may not be much use, as the seed effects are consistent on one computer/install/... only. For instance, the current seed works for me. (pass at 51.79MHz)

ianjfrosst, 2019-11-16:
Ah, I see. Though it might be deterministic. Well, that makes this issue irrelevant. Might make a PR making the seed value more clear for the FPGA docs.

closed 2019-11-16 by ianjfrosst