Spritetm/hadbadge2019_fpgasoc #143: frame buffer problem causes horizontal line problems at specific y positions

mattvenn, 2019-11-28:
**Expected behaviour**

drawing a horizontal line by setting pixels in the frame buffer should result in a single pixel wide line on the display.

**Resulting behaviour**

- Drawing a line at y = 2 works as expected
- Drawing a line at y = 3 results in 2 short lines with a pixel space between them
- Drawing a line at y = 4 results in 2 px wide line in some places
- Drawing a line at y = 5 works as expected
[image]

**To reproduce**

- Clone https://github.com/mattvenn/had-badge-draw-app/tree/linetest
- make and copy to badge
- button up draws a 200 pix horizontal line at y = 2
- button right draws at y = 3 (with glitches as described above)
- button down draws at y = 4 (with glitches as above)
- button left draws at y = 5 which works as expected

shaos, 2023-10-01:
Based on my experience it looks like DMA draws every ODD line of framebuffer twice (and EVEN lines are completely ignored). This is happening with FB only - tiles and sprites look high-res...

shaos, 2023-10-01:
It sounds funny, but I think I've just fixed EVEN lines issue :)

Example with lines from Matt now draw "ghost" short line in case of y=4, but in all other cases lines look correct (I made background blue to count pixels):
[image]

shaos, 2023-10-01:
https://github.com/Spritetm/hadbadge2019_fpgasoc/pull/155

Spritetm, 2023-11-30:
Seems fixed by Shaos' PR.

closed 2023-11-30 by Spritetm