Spritetm/hadbadge2019_fpgasoc #151: soc svf upload failed

w4ilun, 2020-05-18:
After running `make` then `make prog ` in the soc folder, got the following error:

```
Error: tdo check error at line 31012
Error:     READ = 0x0001e00
Error:     WANT = 0x0000100
Error:     MASK = 0x0002100

Time used: 2m56s701ms 
svf file programmed failed
```

Any ideas?

smunaut, 2020-05-18:
You should be using DFU and not openocd / svf.

The production bootloader locks the flash so the `make prog` won't work out of the box, it's only for pre-production / blank units.

w4ilun, 2020-05-18:
Thanks @smunaut! Sorry I forgot to mention I have a blank unit ( made the pcb/badge myself as I wasn't able to attend the conference this year 😢 )